wradlib/wradlib #622: How to plot DBZV-velocity properly (single elevation file H5)

Miron025, 2023-04-05:
I frequently using classic Wetterdienst module for obtaining live radar data and that's ok, working at 100%....
But now I need to open the single sweep data file (H5). It is part of a dataset which is something like: folder12:05 files: file4.h5, file3.h5 etc...Each file represents one elevation so when using classical method as:

import wradlib as wrl
import warnings
warnings.filterwarnings('ignore')
import matplotlib.pyplot as pl
import numpy as np
import xarray as xr
try:
    get_ipython().magic("matplotlib inline")
except:
    pl.ion()
from wradlib.io.xarray_depr import CfRadial, OdimH5

fpath = 'HD/2782021/ras07-vol5minng01_sweeph5onem_dbzv_05-2021082713230200-drs-10488-hd5'
f = wrl.util.get_wradlib_data_file(fpath)
cf1 = OdimH5(f, georef=True)

fig = pl.figure(figsize=(10,10))
cf1['sweep_1'].DBZV.wradlib.plot_ppi(fig=fig, cmap="NWSVel", vmin=-32.13529120559098, vmax=32.13529120559098,)

The results are senseless....even when I'm using a vmin/vmax values for velocity. How to set the file to be plotted with right colors green / red, towards-away from the radar? See the attached image that showing the bad results.

Thanks

[image]

kmuehlbauer, 2023-04-12:
@Miron025 The example is showing the plot of vertical refelctivities (DBZV) not Doppler velocities (eg. VRADH). Is this just by accident?

Miron025, 2023-04-12:
> @Miron025 The example is showing the plot of vertical reflectivities (DBZV) not Doppler velocities (eg. VRADH). Is this just by accident?
[image]

Other files that were provided by DWD is marked as DBZH and they are ok..maybe they made a mistake by sending wrong files...See this link for the zip file containing all data provided by DWD for my research: 
https://ulozto.net/file/SPNKDMKhUjUc/anfrage-drs-tar?_ga=2.254922760.1855310278.1681305415-760771707.1657923839&_gl=1*1q521z5*_ga*NzYwNzcxNzA3LjE2NTc5MjM4Mzk.*_ga_L4S9E2Y7WN*MTY4MTMwNTQxNC43Ny4xLjE2ODEzMDU1MjQuMC4wLjA.

kmuehlbauer, 2023-04-12:
@Miron025 I'm working a lot with DWD CBand Radar data. And what I can tell is that the files with `dbzv` within the name are vertical reflectivities. 

It's indeed misleading that the subfolders are named `_vel`. I'd ask the data provider if they can clarify on that

Miron025, 2023-04-12:
> @Miron025 I'm working a lot with DWD CBand Radar data. And what I can tell is that the files with `dbzv` within the name are vertical reflectivities.
> 
> It's indeed misleading that the subfolders are named `_vel`. I'd ask the data provider if they can clarify on that

I have sent them an email requesting VELOCITY (wind data) and radar data. It is possible that they may have sent the wrong files by mistake. Currently, I am working on official scientific project in the Czech Republic, where a new radar will be installed in southern Moravia. My role in the project involves analyzing low-topped convection radar data, and the data for this particular case are very important. Unfortunately, the official CZ meteorological institute does not support projects like this and does not share their data. I have spent over 15 years working on this project and have terabytes of data and photos showing unique hybrid events. However, it is quite complicated to explain everything here.

kmuehlbauer, 2023-09-07:
Feel free to reopen.

closed 2023-09-07 by kmuehlbauer