wradlib/wradlib #623: Reading and plotting vertical scan from DWD

guidocioni, 2023-04-19:
This may be a stupid question but I was wondering if there's any way to replicate a plot like this 

[image]

I think the data is available in BUFR data here https://opendata.dwd.de/weather/radar/sites/lmax/ but I don't find any example in the documentation that shows if it's possible or not to work with this kind of data.

Thanks :)

guidocioni, 2023-04-19:
> @guidocioni Reading BUFR has been a pain several years ago. That's one reason why wradlib doesn't have reading abilities. In recent years ECMWF has put great effort into the development of [ecCodes](https://github.com/ecmwf/eccodes) which might make this easier nowadays. The major pain still is to install the needed code tables from the weather service which might be needed for decoding.
> 
> If you are up to an adventure I'd gladly assist to implement some prototype. But otherwise there are no immediate plans to implement this in wradlib.

Yeah I do know that BUFR is a pain in the neck, and in the past I had to resort to other libraries like https://github.com/ecmwf/pdbufr, but still the problem persists.

I guess my question was more general: is there any radar product from DWD that contains vertical scans of reflectivity and can be read in `wradlib`? Sorry for not being clear before :)

kmuehlbauer, 2023-09-07:
@guidocioni I'll close this as wont-fix for now.

> guess my question was more general: is there any radar product from DWD that contains vertical scans of reflectivity and can be read in `wradlib`? Sorry for not being clear before :)

Not that I know of. I know DWD measures vertical scans, but they are not part of the open data.

closed 2023-09-07 by kmuehlbauer