wradlib/wradlib #625: Opening RADOLAN RW composite with Dask LocalCluster crashes

Campostrini, 2023-05-07:
Opening an RW Radolan composite with dask by using `chunks={"time": 1}` AND by using a dask.distributed.LocalCluster crashes in a not-so-informative way

#### MCVE Code Sample
First download one file from the DWD
```bash
wget https://opendata.dwd.de/weather/radar/radolan/rw/raa01-rw_10000-2305051550-dwd---bin.bz2
bzip2 -d raa01-rw_10000-2305051550-dwd---bin.bz2
```

Then run the following
```python
import wradlib as wrl
from dask.distributed import Client

client = Client()

dwd_file = 'raa01-rw_10000-2305051550-dwd---bin'
o = wrl.io.open_radolan_dataset(dwd_file, chunks={"time": 1})
print(o.isel(time=0).compute())
```

#### Expected Output
You can obtain the expected output by simply not instantiating a `dask.distributed.Client`
```
<xarray.Dataset>
Dimensions:  (y: 900, x: 900)
Coordinates:
    time     datetime64[ns] 2023-05-05T15:50:00
  * y        (y) float64 -4.658e+06 -4.657e+06 ... -3.76e+06 -3.759e+06
  * x        (x) float64 -5.23e+05 -5.22e+05 -5.21e+05 ... 3.75e+05 3.76e+05
Data variables:
    RW       (y, x) float32 nan nan nan nan nan nan ... nan nan nan nan nan nan
Attributes:
    radarid:         10000
    formatversion:   3
    radolanversion:  2.29.1
    radarlocations:  ['asb', 'boo', 'ros', 'hnr', 'umd', 'pro', 'ess', 'fld',...
    moduleflag:      1
```

#### Problem Description
It raises this error upon calling `.compute()`. Full Traceback omitted for simplicity. I can paste it if needed.
```
TypeError: _radolan_file.__init__() got an unexpected keyword argument 'mode'
```
#### Version

<details><summary>Output of <tt>wrl.show_versions()</tt></summary>

UserWarning: Setuptools is replacing distutils.
  warnings.warn("Setuptools is replacing distutils.")

INSTALLED VERSIONS
------------------
commit: None
python: 3.11.3 | packaged by conda-forge | (main, Apr  6 2023, 08:57:19) [GCC 11.3.0]
python-bits: 64
OS: Linux
OS-release: 5.4.0-135-generic
machine: x86_64
processor: x86_64
byteorder: little
LC_ALL: None
LANG: en_US.UTF-8
LOCALE: ('en_US', 'UTF-8')
libhdf5: 1.14.0
libnetcdf: 4.9.2

xarray: 2023.4.2
pandas: 2.0.1
numpy: 1.24.3
scipy: 1.10.1
netCDF4: 1.6.3
pydap: None
h5netcdf: 1.1.0
h5py: 3.8.0
Nio: None
zarr: None
cftime: 1.6.2
nc_time_axis: None
PseudoNetCDF: None
iris: None
bottleneck: None
dask: 2023.4.1
distributed: 2023.4.1
matplotlib: 3.7.1
cartopy: 0.21.1
seaborn: None
numbagg: None
fsspec: 2023.4.0
cupy: None
pint: None
sparse: None
flox: None
numpy_groupies: None
setuptools: 67.7.2
pip: 23.1.2
conda: None
pytest: None
mypy: None
IPython: None
sphinx: None

wradlib: 1.19.1

</details>

kmuehlbauer, 2023-05-14:
Thanks for raising this. Indeed would it be nice to have the full traceback.

kmuehlbauer, 2023-05-15:
@Campostrini I was able to reproduce. The issue is with `.compute()`. It creates a new Dataset bei _reloading_ the data from the underlying store. For that it has to make sure the data is not tampered with and it want's to open the store with `mode="r"` for that reason.

The radolan backend is read-only so it does not need a special handling of `mode` IMHO, but I can't immediately see if this might be fixed on the xarray side. The less intrusive change is to add a `**kwargs` to `_radolan_file.__init__()` method.

https://github.com/wradlib/wradlib/blob/31f09557195621c8c29dda07d1b4c90aca0c488b/wradlib/io/radolan.py#L1029

kmuehlbauer, 2023-05-15:
The fix is in e7fc8891cee05344098e978c0ad531fb3f8b1419. @Campostrini It would be great if you could apply the suggested change to your installation, and check if this fixes the issue on your side, too.

Campostrini, 2023-05-15:
Great! I suppose the full traceback is no longer needed then. Will implement the suggested changes and come back to you, but I expect for it to fix the issue 😄 Many thanks @kmuehlbauer

kmuehlbauer, 2023-05-15:
> I suppose the full traceback is no longer needed then. Will implement the suggested changes and come back to you, but I expect for it to fix the issue.

Yes, no need for the traceback anymore. Take your time testing this. Just to be on the safe side before getting a new release out.

kmuehlbauer, 2023-05-31:
@Campostrini Did you had the chance to test this. I'd have a release out, if you give a go.

Campostrini, 2023-06-03:
@kmuehlbauer sorry for the delay! Will do this today. Thanks for reminding me

Campostrini, 2023-06-04:
@kmuehlbauer I confirm it works! I have also noticed that the original few lines I provided don't actualy work unless one rewrites them as follows.

```python
import wradlib as wrl
from dask.distributed import Client

if __name__ == '__main__':
    client = Client()

    dwd_file = 'raa01-rw_10000-2305051550-dwd---bin'
    o = wrl.io.open_radolan_dataset(dwd_file, chunks={"time": 1})
    print(o.isel(time=0).compute())
```

Campostrini, 2023-06-04:
@kmuehlbauer I saw you didn't add a test for this (and maybe it's not needed). However, I was wondering if in the next weeks (without urgency) I could try to make a PR to add some tests for this specific case. 

Would that be useful? (It would be my first PR)