wradlib/wradlib #626: Add probability of meteo echoes to output of "classify_echo_fuzzy"

overeem11, 2023-05-24:
At KNMI we use the probability of meteorological echoes (q) as provided by function "classify_echo_fuzzy" of script "clutter.py" to compute uncertainty information of radar QPE. Currently, the function does not output this probability q.

#### MCVE Code Sample

```python
# Your code here
return np.where(q < thresh, True, False), nan_mask
```

#### Expected Output
A tuple of two boolean arrays of same shape as the input arrays The first array boolean array indicates non-meteorological echos based on the fuzzy classification. The second boolean array indicates where all the polarimetric moments had missing values which could be used as an additional information criterion.

#### Problem Description

I suggest to add q, i.e., the membership in meteorological class after combining all variables, as expected output of this function (a tuple of a float array). This can be easily done by adding ", q" to the line with the return statement:

```python
return np.where(q < thresh, True, False), nan_mask, q 
```

#### Version

<details><summary>Output of <tt>wrl.show_versions()</tt></summary>

INSTALLED VERSIONS
------------------
commit: None
python: 3.10.10 (main, Feb  8 2023, 00:00:00) [GCC 12.2.1 20221121 (Red Hat 12.2.1-4)]
python-bits: 64
OS: Linux
OS-release: 6.2.11-100.fc36.x86_64
machine: x86_64
processor: x86_64
byteorder: little
LC_ALL: None
LANG: en_US.utf8
LOCALE: ('en_US', 'UTF-8')
libhdf5: 1.12.1
libnetcdf: 4.8.1

xarray: 2022.6.0
pandas: 1.3.5
numpy: 1.22.0
scipy: 1.8.1
netCDF4: 1.5.8
pydap: None
h5netcdf: 1.1.0
h5py: 3.6.0
Nio: None
zarr: None
cftime: 1.5.2
nc_time_axis: None
PseudoNetCDF: None
rasterio: None
cfgrib: None
iris: None
bottleneck: 1.3.2
dask: 2023.4.0
distributed: None
matplotlib: 3.5.3
cartopy: 0.21.1
seaborn: 0.11.1
numbagg: None
fsspec: 2023.4.0
cupy: None
pint: None
sparse: None
flox: None
numpy_groupies: None
setuptools: 59.6.0
pip: 21.3.1
conda: None
pytest: 6.2.5
IPython: 8.3.0
sphinx: 4.4.0

wradlib: 1.19.1
</details>

kmuehlbauer, 2023-05-24:
@overeem11 Thanks Aart for this feature request.

What I'm thinking now is to just output the probability `q` instead of the mask. That way we could also take `thresh` out of the function. The remaining one-liner to create the boolean array with `thresh` could be run in the user code. Would that be something we could change for wradlib 2.0?

overeem11, 2023-05-24:
Perhaps that for existing users it would be more convenient to keep thresh. Then they only need to add ProbMeteo in the example below and no other modifications are needed:
cmap, nanmask, ProbMeteo = classify_echo_fuzzy(dat,weights=weights,trpz=trpz,thresh=0.3)

But it would also be fine with me to remove thresh. Perhaps this one-liner "np.where(q < thresh, True, False)" could then be added to https://docs.wradlib.org/en/latest/generated/wradlib.clutter.classify_echo_fuzzy.html or, perhaps better, the Tutorials and Examples (https://docs.wradlib.org/en/latest/notebooks/classify/wradlib_fuzzy_echo_classify.html) to help the user applying this clutter filter.

kmuehlbauer, 2023-05-24:
@overeem11 Thanks, as this will break users code anyway, I'm inclined to follow along your suggestion of adding the `cmap` derivation to the docstring as well as the examples. I'll put it on the 2.0 milestone.

kmuehlbauer, 2023-09-07:
@overeem11 I've taken the clean approach of just returning probability instead of cmap. I've added notes to docstring, fuzzy notebook as well as migration guide. This will shortly surface in wradlib 2.0.0-rc.1.

closed 2023-09-07 by kmuehlbauer