wradlib/wradlib #629: Merge Error

Miron025, 2023-06-26:
Hi!

I started to get this error for no reason. I using the same saved code for a very long time:

This part: 

vol = wrl.io.RadarVolume()
for r, v in zip(volume_reflectivity, volume_velocity):
    ds0 = wrl.io.open_odim_mfdataset(r, group="dataset1",
                                     concat_dim="time",
                                     combine="nested",
                                    )
    ds1 = wrl.io.open_odim_mfdataset(v, group="dataset1",
                                     concat_dim="time",
                                     combine="nested",
                                    )

    vol.append(xr.merge([ds0, ds1], combine_attrs="override"))
    vol.sort(key=lambda x: x.time.min().values)

resulted in this error:
MergeError: conflicting values for variable 'elevation' on objects to be combined. You can skip this check by specifying compat='override'.

Or this one: ValueError: invalid literal for int() with base 10: 't1'

If I will use compat=override function the output images are blank or wrong elevation...I tried it with several DWD radar stations....

(My wetterdients version: '0.32.0')

kmuehlbauer, 2023-06-26:
Please provide the output of `wradlib.show_versions()`. 

We are currently in a transition-phase to use xradar as data-io. The issue might be due to that.

Depending on the wradlib/xradar version, specifying `group='sweep_0'` might fix your issue.

Miron025, 2023-07-10:
> Please provide the output of `wradlib.show_versions()`.
> 
> We are currently in a transition-phase to use xradar as data-io. The issue might be due to that.
> 
> Depending on the wradlib/xradar version, specifying `group='sweep_0'` might fix your issue.

Not working it shows this error: ValueError: time already exists as coordinate or variable name.

Miron025, 2023-07-10:
> @Miron025Please provide as much information as you can. This includes complete stacktraces! I'm trying to reproduce this, but I'm missing necessary information (see also above).
My code:
[code.txt](https://github.com/wradlib/wradlib/files/12002621/code.txt)

I have to leave now because storms are approaching our position, so I don't have time to fix it now. I'll get back to it when its over :)

Miron025, 2023-07-10:
I fixed the error by updating wradlib, everything is working now...thanks :)
[image]

closed 2023-07-10 by kmuehlbauer