wradlib/wradlib #633: Quick help: ValueError: cannot reshape array of size 0 into shape (0,newaxis)

Miron025, 2023-07-26:
I loaded 24 h of data from Dresden and still getting this error...maybe some radar images were possibly damaged or improperly loaded. 
Only pm = vol[0] / 5,5 el. can be loaded. But I need to view lower elevations...

ValueError: cannot reshape array of size 0 into shape (0,newaxis)

vol = wrl.io.RadarVolume()
for r, v in zip(volume_reflectivity, volume_velocity):
    ds0 = wrl.io.open_odim_mfdataset(r, group="dataset1",
                                     concat_dim="time",
                                     combine="nested",
                                    )
    ds1 = wrl.io.open_odim_mfdataset(v, group="dataset1",
                                     concat_dim="time",
                                     combine="nested",
                                    )

    vol.append(xr.merge([ds0, ds1], combine_attrs="override"))
    vol.sort(key=lambda x: x.time.min().values)

Error:

vol = wrl.io.RadarVolume()
for r, v in zip(volume_reflectivity, volume_velocity):
    ds0 = wrl.io.open_odim_mfdataset(r, group="dataset1",
                                     concat_dim="time",
                                     combine="nested",
                                    )
    ds1 = wrl.io.open_odim_mfdataset(v, group="dataset1",
                                     concat_dim="time",
                                     combine="nested",
                                    )

    vol.append(xr.merge([ds0, ds1], combine_attrs="override"))
    vol.sort(key=lambda x: x.time.min().values)

Is there any way how to save and view the data before they will be deleted from DWD folders?

Miron025, 2023-08-07:
Any help? Today the same error while loading 4 hours of data from DRS, only vol=0 can be viewed, lower elevations resulted in this annyoing error and making it unable to see the whole radar image (the lowest scans).

kmuehlbauer, 2023-08-15:
Sorry for the delay here. Unfortunately the stacktrace in the above errormessage.txt is not complete. Please use the Bug-report template and provide the necessary information. Thanks!

kmuehlbauer, 2023-09-05:
Please reopen with added information. Thanks.

closed 2023-09-05 by kmuehlbauer