wradlib/wradlib #634: Building the Python wheel

meerfrau, 2023-07-30:
When building the Python wheel against git master (Python 3.10) with
```
python -m build --wheel --no-isolation
python -m installer --destdir="$pkgdir" dist/*.whl
```
the result is an empty bogus file:
```
└── usr
    └── lib
        └── python3.10
            └── site-packages
                └── UNKNOWN-1.20.0.dev11+g17f876c2.dist-info
                    ├── LICENSE.txt
                    ├── METADATA
                    ├── RECORD
                    ├── top_level.txt
                    └── WHEEL
```

kmuehlbauer, 2023-08-15:
Sorry for the delay here. Could you please add more context here, since I can't reproduce this. Which file is empty?

kmuehlbauer, 2023-09-05:
Please reopen with additional information. Thanks.

closed 2023-09-05 by kmuehlbauer