wradlib/wradlib #651: vis.plot_scan_strategy not working when terrain=True

JulianGiles, 2023-10-25:
#### MCVE Code Sample

```python
import wradlib as wrl
import numpy as np

ranges = np.arange(0, 10000, 100)
elevs = np.arange(1, 30, 3)
site = (-28.5, 38.5, 100.0)

wrl.vis.plot_scan_strategy(
        ranges, elevs, site, az=10, terrain=True, # without terrain=True this works fine
    )
```

#### Expected Output

The scan strategy plot with the silhouette of the terrain.

#### Problem Description

I get the following error:
```
Traceback (most recent call last):

  Cell In[10], line 1
    wrl.vis.plot_scan_strategy(

  File ~/mambaforge/envs/wradlib4/lib/python3.11/site-packages/wradlib/vis.py:446 in plot_scan_strategy
    ds = io.get_srtm(

  File ~/mambaforge/envs/wradlib4/lib/python3.11/site-packages/wradlib/io/dem.py:170 in get_srtm
    status_code = download_srtm(filename, path, resolution)

TypeError: download_srtm() takes 2 positional arguments but 3 were given
```

A few times it also loaded something very heavy (about 20 GB) for a few minutes to then crash because of full memory.

#### Version

<details><summary>Output of <tt>wrl.show_versions()</tt></summary>

INSTALLED VERSIONS
------------------
commit: None
python: 3.11.5 | packaged by conda-forge | (main, Aug 27 2023, 03:34:09) [GCC 12.3.0]
python-bits: 64
OS: Linux
OS-release: 5.14.21-150400.24.81-default
machine: x86_64
processor: x86_64
byteorder: little
LC_ALL: None
LANG: de_DE.UTF-8
LOCALE: ('de_DE', 'UTF-8')
libhdf5: 1.14.2
libnetcdf: 4.9.2

xarray: 2023.9.0
pandas: 2.1.1
numpy: 1.26.0
scipy: 1.11.3
netCDF4: 1.6.4
pydap: None
h5netcdf: 1.2.0
h5py: 3.9.0
Nio: None
zarr: None
cftime: 1.6.2
nc_time_axis: None
PseudoNetCDF: None
iris: None
bottleneck: 1.3.7
dask: 2023.9.2
distributed: 2023.9.2
matplotlib: 3.8.0
cartopy: 0.22.0
seaborn: None
numbagg: None
fsspec: 2023.9.2
cupy: None
pint: 0.22
sparse: None
flox: None
numpy_groupies: None
setuptools: 68.2.2
pip: 23.2.1
conda: None
pytest: None
mypy: None
IPython: 8.15.0
sphinx: 7.2.6

wradlib: 2.0.1
/user/jgiles/mambaforge/envs/wradlib4/lib/python3.11/site-packages/_distutils_hack/__init__.py:33: UserWarning: Setuptools is replacing distutils.
  warnings.warn("Setuptools is replacing distutils.")

</details>

kmuehlbauer, 2023-10-30:
@JulianGiles Thanks for the report. This is a somehow tricky issue. 

In general this doesn't try to download the data if it is available. But some tiles (all sea-tiles) are not available at all. The "dumb" downloader tries to fetch those tiles (which are not there anyway) and errors out (the error you are seeing is another regression).

I'll try to whitelist the all-sea tiles in the machinery. It's a bit of handwork involved, so it might take a bit of time. Nevertheless I'm aiming for a release by end of week :crossed_fingers:.

kmuehlbauer, 2023-11-03:
fixed by #653

closed 2023-11-03 by kmuehlbauer