wradlib/wradlib #654: ValueError: cannot reshape array of size 0 into shape (0,newaxis)

Miron025, 2023-11-14:
I am still getting this annoying error even though the same code seems to have been working for a very long time. I have already updated Wradlib, and everything is working fine when downloading live data (near-live data) up to 10-15 minutes from DWD. However, when I try to download a longer period, for example, 130-150 minutes or more, it results in this error. It loads only a vol_0 (5,5 elevation), which is not useful in most cases, especially when my research is focused on low-topped convection...:D

I hope that this time I have provided enough information to solve this problem :)

[example code.zip](https://github.com/wradlib/wradlib/files/13350555/example.code.zip)

kmuehlbauer, 2023-11-15:
@Miron025 Your code works for me with:

```python
print(wrl.show_versions())
print(f"xradar: {xradar.__version__}")
```

```
INSTALLED VERSIONS
------------------
commit: None
python: 3.11.4 | packaged by conda-forge | (main, Jun 10 2023, 18:08:17) [GCC 12.2.0]
python-bits: 64
OS: Linux
OS-release: 6.1.0-13-amd64
machine: x86_64
processor: 
byteorder: little
LC_ALL: None
LANG: de_DE.UTF-8
LOCALE: ('de_DE', 'UTF-8')
libhdf5: 1.14.2
libnetcdf: 4.9.2

xarray: 2023.7.0
pandas: 2.0.3
numpy: 1.25.1
scipy: 1.11.1
netCDF4: 1.6.5
pydap: None
h5netcdf: 1.2.0
h5py: 3.10.0
Nio: None
zarr: None
cftime: 1.6.2
nc_time_axis: None
PseudoNetCDF: None
iris: None
bottleneck: None
dask: 2023.7.1
distributed: 2023.7.1
matplotlib: 3.7.2
cartopy: 0.22.0
seaborn: None
numbagg: None
fsspec: 2023.6.0
cupy: None
pint: 0.22
sparse: None
flox: None
numpy_groupies: None
setuptools: 68.0.0
pip: 23.2.1
conda: None
pytest: None
mypy: None
IPython: 8.14.0
sphinx: None

wradlib: 1.19.2
xradar: 0.3.0
```

Output:

```python
print(vol)
<wradlib.RadarVolume>
Dimension(s): (sweep: 6)
Elevation(s): (5.5, 4.5, 3.5, 2.5, 1.5, 0.5)
```

```python
print(vol[0])
<xarray.Dataset>
Dimensions:            (azimuth: 360, range: 720, time: 25)
Coordinates:
  * azimuth            (azimuth) float64 0.5 1.5 2.5 3.5 ... 357.5 358.5 359.5
  * range              (range) float32 125.0 375.0 625.0 ... 1.796e+05 1.799e+05
    elevation          (azimuth) float64 dask.array<chunksize=(360,), meta=np.ndarray>
    rtime              (time, azimuth) datetime64[ns] dask.array<chunksize=(1, 360), meta=np.ndarray>
    sweep_mode         <U20 'azimuth_surveillance'
    longitude          float64 11.18
    latitude           float64 52.16
    altitude           float64 185.2
  * time               (time) datetime64[ns] 2023-11-15T04:50:35.349500160 .....
Data variables:
    DBZH               (time, azimuth, range) float32 dask.array<chunksize=(1, 360, 720), meta=np.ndarray>
    sweep_number       (time) int64 0 0 0 0 0 0 0 0 0 0 ... 0 0 0 0 0 0 0 0 0 0
    prt_mode           (time) <U7 'not_set' 'not_set' ... 'not_set' 'not_set'
    follow_mode        (time) <U7 'not_set' 'not_set' ... 'not_set' 'not_set'
    sweep_fixed_angle  (time) float64 5.499 5.499 5.499 ... 5.499 5.499 5.499
    VRADH              (time, azimuth, range) float32 dask.array<chunksize=(1, 360, 720), meta=np.ndarray>
Attributes:
    fixed_angle:  5.4986572265625
```

kmuehlbauer, 2023-11-15:
@Miron025 It looks like you are on an older version of xradar. wradlib moved all xradar reading capabilities over to xradar-package and only kept a shallow compatibility layer for the 1.19 code line. Please update xradar.

kmuehlbauer, 2023-12-01:
@Miron025 Did updating xradar help with your problem?

kmuehlbauer, 2024-02-05:
@Miron025 Is this still an issue for you? Please reopen with more details.

closed 2024-02-05 by kmuehlbauer