wradlib/wradlib #661: Radolan 1100x900 extent

JRoehrig, 2023-12-19:
Hello,

first of all, thank you for providing the package (under MIT license), I am starting to use it in my hydrology lectures.

When I create a tif-file from radolan 1100x900 found in [DWD-CDC](https://opendata.dwd.de/climate_environment/CDC/grids_germany/5_minutes/radolan/reproc/2017_002/bin/2001/) the resulting extent does not coincide with the expected (-443.46216692185595, -4758.644724265573, 456.53783307814405, -3658.644724265573). I cannot find my error.

#### MCVE Code Sample

```python
import os
import wradlib as wrl
w_dir = os.path.join(wrl.util.get_wradlib_data_path(), 'radolan/check')
fn = "5_minutes/radolan/reproc/2017_002/bin/2001/YW2017.002_20010101/raa01-yw2017.002_10000-0101010450-dwd---bin"
fn = wrl.util.get_wradlib_data_file(fn)
data_raw, meta = wrl.io.read_radolan_composite(fn)
crs_rad = wrl.georef.create_osr("dwd-radolan")
xy = wrl.georef.get_radolan_grid(1100, 900, crs=crs_rad)  # bounding box: OK. mode will be radolan as per default, which evaluates LowerLeft corners 
data, xy = wrl.georef.set_raster_origin(data_raw, xy, "upper")  # flipud: OK. now xy[0,0,:] is UpperLeft, ready for GeoTransform
ds = wrl.georef.create_raster_dataset(data, xy, crs=crs_rad)  # problem? xy will be shifted here by half a pixel, as if it was centred
wrl.io.write_raster_dataset(os.path.join(w_dir, 'raa01-yw2017.002_10000-0101010450_1500_900_wradlib_1.tif'), ds)
print(ds.GetGeoTransform())
ds = None
```
#### Output
(-443.96216692185595, 1.0, 0.0, -3659.144724265573, 0.0, -1.0)

#### Expected Output
(-443.46216692185595, 1.0, 0.0, -3658.644724265573, 0.0, -1.0) according to the documentation (below)

Coordinates for 1100km x 900km grid (from doc in wradlib.georef.rect.get_radolan_grid())

| Coordinate |   lon     |     lat    |     x     |     y     |
|:-----------|:----------|:-----------|:----------|:----------|
| LowerLeft  |  4.6759E  |  46.1929N  | -443.4622 | -4758.645 |
| UpperRight | 17.1128E  |  55.5342N  |  456.5378 | -3658.645 |

#### Problem Description
*xy = wrl.georef.get_radolan_grid(1100, 900, crs=crs_rad)* returns the bounding box of the raster. It is not centred, i.e., xy  is ready to be used as gt[0] and gt[3] in SetGeoTransform. But in *raster.create_raster_dataset(data, coords, \*, crs=None, nodata=-9999)*  the raster is being shifted by half a pixel in 

    if data.shape[-2:] == coords.shape[0:2]:  # this evaluates to True
        upper_corner_x = coords[0, 0, 0] - x_ps / 2   # shifting half a pixel. I guess this should not be so 
        upper_corner_y = coords[0, 0, 1] - y_ps / 2
    else:
        upper_corner_x = coords[0, 0, 0]  # I guess this should be the case
        upper_corner_y = coords[0, 0, 1]

My workaround is to shift xy to the center after *xy = wrl.georef.get_radolan_grid(1100, 900, crs=crs_rad)* or to use
xy = wrl.georef.get_radolan_grid(1100, 900, crs=crs_rad, mode='center') instead of the (default) mode='radolan'. I am not sure it is intended to be so.

#### Version

<details><summary>Output of <tt>wrl.show_versions()</tt></summary>

wradlib: 2.0.3

</details>

JRoehrig, 2023-12-19:
Ah sorry, I just found the issue # 499 fixed in https://github.com/wradlib/wradlib/pull/502.

I also just checked  https://docs.wradlib.org/en/latest/notebooks/fileio/gis/raster_data.html. Maybe it is worth mentioning this issue there in the cell 7 - Step 2b: (output with GDAL):

```python 
# This is the RADOLAN projection
proj_osr = wrl.georef.create_osr("dwd-radolan")

# Get projected RADOLAN coordinates for cell centers  -> CHANGED
xy_raw = wrl.georef.get_radolan_grid(900, 900, mode='center')
```

kmuehlbauer, 2023-12-19:
@JRoehrig Glad you find it useful. And, you've already found the solution. Nevertheless, I'll answer here.

Yes, you're absolutely right that we should improve the docs and notebooks here for better visibility. Contributions welcome.

It's with the `create_raster_dataset` function. It assumes pixel-center coordinates if data.shape and coords.shape correspond. In this case data and coords shape are 1100,900 assuming pixel-center coordinates. 

You can use `mode="center"` in the call to `get_radolan_grid` to fix this.

Also be aware that `get_radolan_grid` with `mode="radolan"` does only return the lower left pixel values of each grid box and no upper right values. So bounding box can't be retrieved immediately. `mode="edge"` would output also the upper right coordinates with shape (1101, 901).

JRoehrig, 2023-12-19:
I got it. I was wondering when data.shape[-2:] != coords.shape[0:2] in *raster.create_raster_dataset()*  . This is for mode='edge'.
Thank you.

closed 2023-12-19 by JRoehrig