wradlib/wradlib #668: AttributeError: 'NoneType' object has no attribute 'TransformPoints' on data.wrl.georef.georeference() function

panagiotisncm, 2024-03-10:
AttributeError: 'NoneType' object has no attribute 'TransformPoints'

#### Problem Description

While trying to run the [wradlib example](https://docs.wradlib.org/en/latest/notebooks/visualisation/plot_ppi.html) on input [8] and specifically while running the second line "img = img.wrl.georef.georeference(crs=epsg)" i always get an attributte error and specifically: 
 
**AttributeError: 'NoneType' object has no attribute 'TransformPoints'**

The specific error can also be found in other examples such as:

1. [Beam blockage](https://docs.wradlib.org/en/latest/notebooks/beamblockage/beamblockage.html): Input [4], line 3 and specifically the wrl.georef.spherical_to_proj command.
2. [Radar Scan Strategy](https://docs.wradlib.org/projects/old-docs/en/1.9.0/notebooks/visualisation/wradlib_plot_scan_strategy.html): When trying to use the 'terrain=True' argument in 'wrl.vis.plot_scan_strategy' function.

In all three issues above i get the exact same error even when using other datasets instead of the ones provided at the examples.

#### Version

- Python version: 3.11.8
- wradlib 2.0.3
- numpy 1.26.4
- scipy 1.12.0
- matplotlib 3.8.3
- xarray 2024.2.0
- xradar 0.4.3

kmuehlbauer, 2024-03-10:
@panagiotisncm Thanks for raising this.

Please check if you've installed `PROJ` in your environment and please check if the environment variables `PROJ_LIB`  or `PROJ_DATA` (depending on your `PROJ` version) are pointing to the right locations 

You also might need to set environment variable `PROJ_NETWORK=ON` to enable automatic download of the necessary transformation data. 

See https://proj.org/en/9.4/resource_files.html for more information.

panagiotisncm, 2024-03-10:
I have installed the following:

- proj 0.2.0
- proj-data 1.17
- pyproj 3.6.1 

i also set my variables as seen below:

`
import os
os.environ["PROJ_DATA"] = "/home/p-ioannidis/anaconda3/envs/Projects/share/proj/"
os.environ["PROJ_LIB"] = "/home/p-ioannidis/anaconda3/envs/Projects/share/proj/"
os.environ["PROJ_NETWORK"] = 'ON'
`

But still the same error persist.

kmuehlbauer, 2024-03-10:
Thanks for checking. I`m assuming the PROJ version is 9.2.0, right?

As all necessary data should be available (proj-data is installed) it should work.

Can you please check your GDAL version and if GDAL_DATA variable is set to the according `.../share/gdal` folder?

Can you also post the output of `conda list` from your environment.

panagiotisncm, 2024-03-10:
The issue seems to have been resolved after setting the GDAL_DATA variable to the correct path. 

@kmuehlbauer thank you very much for your assistance!

kmuehlbauer, 2024-03-10:
@panagiotisncm Glad you got it working. Thanks for providing detailed descriptions, this helps a lot in getting to the root cause. I'll close the issue.

closed 2024-03-10 by kmuehlbauer