wradlib/wradlib #677: dp.unfold_phi is not working - wradlib 2.0.1

saranya00-s, 2024-09-02:
I was using the function unfold_phi for correcting differential phase profile. But I am getting the same profile as output which is ranging from -180 to 180 degrees. Previous version of wradlib was giving output Phidp profile with min and max values 0 and 360 respectively even when there is no folding.
 
code for unfolded differential phase calculation:
_Phi_unf=wradlib.dp.unfold_phi(PHIDP,RHO, width=5, copy=False)_

Is this the version issue?

kmuehlbauer, 2024-09-02:
@saranya00-s Thanks for the report. Can you create a [MCVE](https://stackoverflow.com/help/minimal-reproducible-example)? This will help to speed up the process. Thanks!

saranya00-s, 2024-09-02:
@kmuehlbauer Sure. Here is the sample code

import wradlib
from wradlib.util import get_wradlib_data_file
import matplotlib.pyplot as plt
  

rhofile = get_wradlib_data_file('netcdf/TAG-20120801-140046-02-R.nc')
phifile = get_wradlib_data_file('netcdf/TAG-20120801-140046-02-P.nc')

rho, attrs_rho = wradlib.io.read_edge_netcdf(rhofile)
phi, attrs_phi = wradlib.io.read_edge_netcdf(phifile)

phiunf=wradlib.dp.unfold_phi(phi, rho, width=5, copy=False)

# Plotting raw Phidp and unfolded Phidp along azimuth 10 deg
plt.figure()
plt.plot(phi[10,:],color="r")
plt.plot(phiunf[10,:],color="b")
plt.legend(('Phidp Raw', 'Phidp Unfolded'),
           loc='upper right')

kmuehlbauer, 2024-09-02:
It seems it works as expected:

```python
# Plotting raw Phidp and unfolded Phidp along azimuth 10 deg
plt.figure()
plt.plot(phi[10,:], color="r", label="Phidp Raw")
plt.plot(phi[225,:], color="r", ls=":")
plt.plot(phiunf[10,:],color="b", label='Phidp Unfolded')
plt.plot(phiunf[225,:],color="b", ls=":")
plt.legend(loc='upper right')
```

saranya00-s, 2024-09-02:
Is the correction is applied only to folded profiles. Unfolded phase at 10 deg azimuth is ranging to -20 degrees and not in the expected range (0,360).

saranya00-s, 2024-09-02:
> Your actual data range is (-180, 180) and it is unfolded to (-180, 180 +
> X) wherever X is what your actual data needs.

Is it meaningful phase ranging from -180 to 180+X

kmuehlbauer, 2024-09-02:
I think, that depends on your data. It will auto-detect the locations where folding occurs and unfold them. This automatically extends your data range on the upper border. As the algorithm has no knowledge of your data (eg. system differential phase) beside that data range is (-180deg, 180deg) (which is a very common format for phase data) the unfolding algorithm just handles folding over the +180deg. If your data would be in (0deg, 360eg) the algorithm would have (0, 360 + X) as output.

saranya00-s, 2024-09-03:
Thank you for looking into the problem.

kmuehlbauer, 2024-09-03:
I'm closing this. If you have any further questions, we can reopen.

closed 2024-09-03 by kmuehlbauer