wradlib/wradlib #678: Projection mismatch in RADOLAN reader?

guidocioni, 2024-09-05:
Today I was trying to plot some data from the RADOLAN RY composite. As far as I'm understand this product should be the closest to 5-minutes precipitation rate estimation from reflectivity (after some quality correction).

I'm using the `xarray` reader by reading a bunch of files. 
```python
ds = xr.open_mfdataset('radar_data/raa01-ry_10000-*-dwd---bin', engine='radolan')
```
This automatically reads the coordinates and the time (which I suppose should be in UTC). 

After I'm plotting with 

```python
tile_source = cimgt.MapboxTiles(access_token=os.environ['MAPBOX_KEY'], map_id='dark-v10')

fig, axis = plt.subplots(1, 1,
                         subplot_kw=dict(projection=tile_source.crs),
                         figsize=(15, 15))
axis.set_extent([8.38, 10.78, 53., 54.17],  crs=ccrs.PlateCarree())
axis.add_image(tile_source, 8, interpolation='spline36')

x2d, y2d = np.meshgrid(ds.x, ds.y)
data = ds.sel(time="2024-09-04 04:15", method='nearest')['RY']

```

kmuehlbauer, 2024-09-05:
@guidocioni How does the image look like, if you use plain wradlib/cartopy/xarray plotting along  https://docs.wradlib.org/en/latest/notebooks/fileio/radolan/radolan_quickstart.html?

The features would have to be added to see the borders/rivers in der image.

kmuehlbauer, 2024-09-05:
Do you know what transformation engine matplotlib uses under the hood?

guidocioni, 2024-09-05:
> Do you know what transformation engine matplotlib uses under the hood?

isn't cartopy taking care of that? I'd suppose this to be quite mature as feature because I used the `transform` argument before and never noticed errors.
Maybe the original map projection I'm specifying doesn't coincide with the original one?

guidocioni, 2024-09-05:
to be fair I'm also assuming the projection used by Kachelmann is 100% correct, which may not be true

guidocioni, 2024-09-06:
Hey @kmuehlbauer I have some updates but you'll need to help me understand what is going on when using the readers (included the xarray one) because there is definitely something weird going on. 
First of all, as you suggested, I tried to use the default reader

```python
data, attrs = wrl.io.read_radolan_composite("radar_data/raa01-ry_10000-2409040415-dwd---bin")
radolan_grid_xy = wrl.georef.get_radolan_grid(900, 900)
data = np.ma.masked_equal(data, -9999) * 20

fig = plt.figure(figsize=(10, 8))
ax = plt.axes(projection=ccrs.Stereographic(
    true_scale_latitude=60.0, central_latitude=90.0, central_longitude=10.0
))
ax.set_extent([5, 15, 47., 54.17], crs=ccrs.PlateCarree())
x = radolan_grid_xy[:, :, 0]
y = radolan_grid_xy[:, :, 1]
pm = ax.pcolormesh(x, y, data, cmap=cmap, norm=norm)
ax.add_feature(cfeature.BORDERS.with_scale('10m'), edgecolor='black', zorder=5)
ax.add_feature(cfeature.COASTLINE.with_scale('10m'), edgecolor='black', zorder=5)
```

Note that I'm choosing the figure `projection` to coincide with the RADOLAN one. However, it seems the coordinates obtained like this are wrong because I'm getting an empty plot.

[image]

If I remove the call to the set extents then the radar data is plotted, but not the geo features, which means there is some mismatch in the `x` and `y`. 

[image]

Then, I saw there is a `wgs84` option when reading the coordinates, so I tried to use that

```python
data, attrs = wrl.io.read_radolan_composite("radar_data/raa01-ry_10000-2409040415-dwd---bin")
data = np.ma.masked_equal(data, -9999) * 20
radolan_grid_xy = wrl.georef.get_radolan_grid(900, 900, wgs84=True)
fig = plt.figure(figsize=(15, 15))
ax = plt.axes(projection=ccrs.Stereographic(
    true_scale_latitude=60.0, central_latitude=90.0, central_longitude=10.0
))
ax.set_extent([5, 15, 47., 54.17], crs=ccrs.PlateCarree())
x = radolan_grid_xy[:, :, 0]
y = radolan_grid_xy[:, :, 1]
pm = ax.pcolormesh(x, y, data, cmap=cmap, norm=norm, transform=ccrs.PlateCarree())
cb = fig.colorbar(pm, shrink=0.2)
ax.add_feature(cfeature.BORDERS.with_scale('10m'), edgecolor='black', zorder=5)
ax.add_feature(cfeature.COASTLINE.with_scale('10m'), edgecolor='black', zorder=5)
ax.set_extent([8.38, 10.78, 53., 54.17], crs=ccrs.PlateCarree())
states_provinces = cfeature.NaturalEarthFeature(
        category='cultural',
        name='admin_1_states_provinces_lines',
        scale='10m',
        facecolor='none')
ax.add_feature(states_provinces, edgecolor='black', zorder=5)
```

Note that now the figure has the same projection, **but the data not anymore so I have to use the `transform` option when calling `pcolormesh`.**
This surprisingly works! I get the right results, which are clearly the same one shown in the kachelmann picture (i also checked other sources and this seems to be correct).

[image]

If I try to do the same using directly the xarray reader (`ds = wrl.io.open_radolan_dataset("radar_data/raa01-ry_10000-2409040415-dwd---bin"`) I get the wrong result

[image]

So, there is something wrong going on in the reprojection when using the xarray backend (plus some error that I cannot understand when just reading the original grid). It seems this feature should be buried somewhere inside `xradar` but I cannot really find where this reprojection is done!

Is there any way to use the `wgs84` option when calling the `xarray` reader?

kmuehlbauer, 2024-09-06:
So, it seems, cartopy is to blame, or how it is handled. If I create the 2D mesh coordinates in WGS84 beforehand (as you did it too), everything looks nice:

```
x2d, y2d = np.meshgrid(ds.x, ds.y)
lon, lat = wrl.georef.reproject(x2d, y2d, src_crs=wrl.georef.create_osr("dwd-radolan-sphere"), trg_crs=wrl.georef.get_default_projection())

data = ds.sel(time="2024-09-04 04:15", method='nearest')['RY']
data = data.where(data>0)

c = axis.pcolormesh(
    lon,
    lat,
    data.values,
    transform=ccrs.PlateCarree(),
    cmap="turbo",
)
```
[image]

It seems that somehow the Cartopy transform using `Stereographic` is not working as expected

guidocioni, 2024-09-06:
> So, it seems, cartopy is to blame, or how it is handled. If I create the 2D mesh coordinates in WGS84 beforehand (as you did it too), everything looks nice:
> 
> ```
> x2d, y2d = np.meshgrid(ds.x, ds.y)
> lon, lat = wrl.georef.reproject(x2d, y2d, src_crs=wrl.georef.create_osr("dwd-radolan-sphere"), trg_crs=wrl.georef.get_default_projection())
> 
> data = ds.sel(time="2024-09-04 04:15", method='nearest')['RY']
> data = data.where(data>0)
> 
> c = axis.pcolormesh(
>     lon,
>     lat,
>     data.values,
>     transform=ccrs.PlateCarree(),
>     cmap="turbo",
> )
> ```
> 
> ![RY-hamburg](https://private-user-images.githubusercontent.com/5821660/365121084-e3b7a21b-6ef9-41f4-a0c5-cc20b801401b.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjU2MTY4MTEsIm5iZiI6MTcyNTYxNjUxMSwicGF0aCI6Ii81ODIxNjYwLzM2NTEyMTA4NC1lM2I3YTIxYi02ZWY5LTQxZjQtYTBjNS1jYzIwYjgwMTQwMWIucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDkwNiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA5MDZUMDk1NTExWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZmQ0MjZmNTU1M2Q3MjJmMTkyMTFhYzFhYTczNThkZjU4NzdmYjc5ZmIyMDIxYzYzNWY3MGEwNzlmMGM5Zjc1MSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.8kuPaqPT2nnrDxbjud_E0KMu0-Zzlczx6roqsUQy6_o)
> 
> It seems that somehow the Cartopy transform using `Stereographic` is not working as expected

Ah ok. To be fair it doesn't surprise me because I had all sorts of issues with cartopy.
However, given that it is described on the `wradlib` doc, I thought many people have used `cartopy` and `wradlib` to plot RADOLAN data...right? How am I the first one discovering this? :) Or is this a corner case?

kmuehlbauer, 2024-09-06:
Yes, probably, we would have to revisit that in the docs.

guidocioni, 2024-09-06:
For the moment I'm manually replacing the coordinates that I'm using for the plot without transforming.
Should we open an issue in cartopy?

kmuehlbauer, 2024-09-06:
@guidocioni A cartopy-only reproducer would be nice, yes.

kmuehlbauer, 2024-09-07:
@guidocioni Could you please check this out:

```python
map_proj = ccrs.Stereographic(
    true_scale_latitude=60.0, central_latitude=90.0, central_longitude=10.0, globe=ccrs.Globe(ellipse="sphere")
)
```

guidocioni, 2024-09-07:
> ```python
> globe=ccrs.Globe(ellipse="sphere")
> ```

Nice catch! Yes it does work like this :) 
Much better than the workaround I implemented. 

So there's definitely something to add to the doc :) 

If you're curious I noticed this because I was trying to compare flight tracks with radar data to see how planes deviate in case of severe weather, so a precise localization was really important. Here is the animation I was able to create based on the data. As soon as a thunderstorm arrives over Hamburg airport you can see the planes going into a hold over `RARUP` (that's a waypoint name).

https://github.com/user-attachments/assets/0973108c-d51f-4950-b491-8e935d6326bc

Thanks again for the help @kmuehlbauer, I will close this.

closed 2024-09-07 by guidocioni

kmuehlbauer, 2024-09-07:
@guidocioni Impressive animation :+1: I'm wondering how many planes can be on hold there, before they get diverted to other airports?

And yes, changes need to be made to the docs, definitely. Especially this one is overlooked quite easy.