wradlib/wradlib #700: Standardize documentation for xarray-based xradar data accessors

d-siden, 2025-03-10:
Hi,

I'd like to request a more standardized approach on how the xradar package modifies the native xarray _open_dataset()_ function parameters.

E.g.: [xarray docs](https://docs.xarray.dev/en/stable/generated/xarray.open_dataset.html#xarray-open-dataset) makes the parameters of the function clearly listed at the page, while [xradar docs](https://docs.openradarscience.org/projects/xradar/en/stable/notebooks/Rainbow.html#) shows examples of using the already present open_dataset() function for each of the modified engines.

I miss the standard documentation, because I make heavy usage of help shortcut in my IDE, and it shows documentation text on-screen, if it's are available, such as in the wradlib package. The image attached below shows that there was no way for me to guess the "rainbow" engine should be used, and the IDE highlights the _group_ parameter in yellow because it's just not present in docs, although the code works fine.

I'd find useful if we had a similar mode to introduce the expanded arguments for the function in the documentation, to have a better sight of the new engines.

Also, together with the _engine_ parameter, could the docs clarify how to use the _group_ parameter. I heavily suspect this refers to sweep elevations, but I'd really prefer the docs to make it clear.

Thanks in advance.

[image]

kmuehlbauer, 2025-03-21:
@d-siden Sorry for the delay. Is this more of an issue for `xradar`? Or does the same issues apply for the remaining engine (`radolan`)?

d-siden, 2025-03-21:
I'd say this is an issue for `xradar`, as I haven't used the remaining engines.

kmuehlbauer, 2026-10-25:
@d-siden I'll close this, please open a follow-up issue at xradar. Thanks!

closed 2026-10-25 by kmuehlbauer