wradlib/wradlib #701: TypeError: togrid() missing 1 required positional argument: 'data'

aladinor, 2025-03-30:
I am trying to reproduce the gridding and composite workflow as shown [here](https://www2.meteo.uni-bonn.de/data/radar/prepep2025/gridding-compositing).  This is the code I am using

#### MCVE Code Sample
```python
import xradar as xd
import fsspec
import matplotlib.pyplot as plt
import wradlib as wrl
import xarray as xr
import numpy as np

def download_s3file(s3_path: str)-> str:
    return fsspec.open_local(
            f"simplecache::s3://{s3_path}",
            s3={"anon": True},
            filecache={"cache_storage": "."},
        )

def bbox(*args):
    """Get bounding box from a set of radar bin coordinates"""
    xy = np.array(
        [
            [
                arg.x.min().values,
                arg.x.max().values,
                arg.y.min().values,
                arg.y.max().values,
            ]
            for arg in args
        ]
    )
    xmin = xy[..., 0].min()
    xmax = xy[..., 1].max()
    ymin = xy[..., 2].min()
    ymax = xy[..., 3].max()

    return xmin, xmax, ymin, ymax

if __name__ == '__main__':

    fs = fsspec.filesystem("s3", anon=True)
    year = 2024
    car_files = ['s3://' + f for f in
                 sorted(fs.glob(f"s3-radaresideam/l2_data/{year}/08/09/Carimagua/CAR*"))
                 ]
    gua_files = ['s3://' + f for f in
                 sorted(fs.glob(f"s3-radaresideam/l2_data/{year}/08/09/Guaviare/GUA*"))
                 ]

    dtree_car = xd.io.open_iris_datatree(
        download_s3file(car_files[7])
        ).xradar.georeference()
        
    dtree_gua = xd.io.open_iris_datatree(
        download_s3file(gua_files[3])
    ).xradar.georeference()

    ctree = xr.DataTree.from_dict(
        {
            "CAR": dtree_car,
            "GUA": dtree_gua
        }
    )

    proj_utm = wrl.georef.epsg_to_osr(4326)
    gtree = xr.DataTree()

    xmin, xmax, ymin, ymax = bbox(*[ctree[f"{radar}/sweep_0"].ds for radar in ctree.children])

    x = np.linspace(xmin, xmax + 1000.0, 1000)
    y = np.linspace(ymin, ymax + 1000.0, 1000)

    cart = xr.Dataset(coords={"x": (["x"], x), "y": (["y"], y)})

    for radar in ctree.children:
        print(radar)
        ctree[f"{radar}/sweep_0"].ds = ctree[f"{radar}/sweep_0"].ds.wrl.georef.georeference(crs=proj_utm)
        ctree[f"{radar}/sweep_0"].ds = ctree[f"{radar}/sweep_0"].ds.assign(
            QUAL=ctree[f"{radar}/sweep_0"].ds.range.wrl.qual.pulse_volume(250.0, 0.9).expand_dims(
                azimuth=ctree[f"{radar}/sweep_0"].ds.azimuth))
        ra, si = ctree[f"{radar}/sweep_0"].ds.DBZH.wrl.trafo.idecibel().wrl.zr.z_to_r_enhanced()
        ctree[f"{radar}/sweep_0"].ds = ctree[f"{radar}/sweep_0"].ds.assign(RAIN=ra)
        gridded = ctree[f"{radar}/sweep_0"].ds.wrl.comp.togrid(
            cart,
            radius=180000.0,
            center=(ctree[f"{radar}/sweep_0"].ds.y.mean(), ctree[f"{radar}/sweep_0"].ds.x.mean()),
            # location of radar in lat/lon
            interpol=wrl.ipol.Nearest,  # scipy KDTree based
        )
    gtree[f"{radar}/sweep_0"] = gridded

```

#### Problem Description
and this is the the output

``` python
CAR
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
File <timed exec>:4

File [~/mambaforge/envs/raw2zarr-dev/lib/python3.12/site-packages/wradlib/comp.py:268](http://localhost:8888/lab/tree/notebooks/~/mambaforge/envs/raw2zarr-dev/lib/python3.12/site-packages/wradlib/comp.py#line=267), in CompMethods.togrid(self, *args, **kwargs)
    266     return togrid(self, *args, **kwargs)
    267 else:
--> 268     return togrid(self._obj, *args, **kwargs)

File [~/mambaforge/envs/raw2zarr-dev/lib/python3.12/functools.py:912](http://localhost:8888/lab/tree/notebooks/~/mambaforge/envs/raw2zarr-dev/lib/python3.12/functools.py#line=911), in singledispatch.<locals>.wrapper(*args, **kw)
    908 if not args:
    909     raise TypeError(f'{funcname} requires at least '
    910                     '1 positional argument')
--> 912 return dispatch(args[0].__class__)(*args, **kw)

TypeError: togrid() missing 1 required positional argument: 'data'
```

#### Version

<details><summary>Output of <tt>wrl.show_versions()</tt></summary>

INSTALLED VERSIONS
------------------
commit: None
python: 3.12.9 | packaged by conda-forge | (main, Mar  4 2025, 22:48:41) [GCC 13.3.0]
python-bits: 64
OS: Linux
OS-release: 6.8.0-55-generic
machine: x86_64
processor: x86_64
byteorder: little
LC_ALL: None
LANG: en_US.UTF-8
LOCALE: ('en_US', 'UTF-8')
libhdf5: 1.14.4
libnetcdf: 4.9.2

xarray: 2025.3.0
pandas: 2.2.3
numpy: 2.1.3
scipy: 1.15.2
netCDF4: 1.7.2
pydap: None
h5netcdf: 1.6.1
h5py: 3.12.1
zarr: 3.0.6
cftime: 1.6.4
nc_time_axis: None
iris: None
bottleneck: 1.4.2
dask: 2025.3.0
distributed: 2025.3.0
matplotlib: 3.10.1
cartopy: 0.24.0
seaborn: None
numbagg: None
fsspec: 2025.3.0
cupy: None
pint: 0.24.4
sparse: None
flox: None
numpy_groupies: None
setuptools: 75.8.2
pip: 25.0.1
conda: None
pytest: 8.3.5
mypy: None
IPython: 9.0.2
sphinx: None

wradlib: 2.4.0
</details>

Any ideas what am I doing wrong @kmuehlbauer ?

kmuehlbauer, 2025-03-30:
Thanks for raising this, @aladinor. I'll have a check in this tomorrow.

kmuehlbauer, 2025-03-30:
@aladinor I'm sorry, I missed to push a last minute fix upstream to wradlib. Please check your workflow with #702. The issue was, that `comp.togrid` was not fit for `xr.Dataset` yet.

Sorry for any inconvenience!

closed 2025-03-31 by kmuehlbauer

aladinor, 2025-03-31:
Thanks @kmuehlbauer for your help!! : )

kmuehlbauer, 2025-03-31:
Yeah, this fell through the cracks during the conference. :grimacing: 

BTW, there is another gridder upcoming in #691 by @egouden.

PR #702 (2025-03-31): FIX: make comp.togrid work for Datasets
closes #701