wradlib/wradlib #712: integrate notebooks

egouden, 2025-08-25:
I see several reasons to integrate the notebooks in the main wradlib repository:

- integrate notebook in testing
- avoid breaking change
- better visibility for users
- follow xradar strategy

kmuehlbauer, 2025-08-25:
Thanks Edouard for creating this issue.

I agree. The root causes to have a dedicated notebooks-repo do not exist any more and the handling now is a bit of a pain. We have brought the docs back already in v2.0.0 for the same reason and I think it's time to also include the notebooks.

This might look like an easy task, but we should discuss this in more detail. 

What I would like bring into the discussion:

- use [MyST]](https://mystmd.org/) markdown for writing notebooks instead of regular notebooks
- drop Python intro, instead link to other resources (NumPy, SciPy etc)
- restructure the tutorials (eg. dedicated Getting started section with low barrier simple examples)
- drop backend specific notebooks which are already handled in xradar
- add a gallery reel

kmuehlbauer, 2025-08-26:
Can you give a bit more detail how you envision the pipeline where the notebook rendering does  not run on ReadTheDocs.

The current pipeline works quite well and if we have the new notebooks in wradlib main, we can still use it. All source files in one place, only the rendered output over at wradlib-notebooks.

I can add a PR with my suggestion later this week

kmuehlbauer, 2025-08-29:
@egouden I'm on it, but this will take a bit more time as expected. I'm following multiple path's now, and hope to have a PR ready early next week. Thanks for your patience.