wradlib/wradlib #718: pyproj vs gdal osr

kmuehlbauer, 2025-09-22:
Back in the days we've implemented wradlib's projection handling using gdal osr. At that time, this was at least for the Python side of things the most accessible solution.

Now it's time to re-assess the situation. `pyproj` Python package is mature enough, to enhance our projection handling alongside our gdal osr based implementation. 

My suggestion would be to refactor our internals to work with either [pyproj.CRS](https://pyproj4.github.io/pyproj/stable/api/crs/crs.html) or [gdal.osr](https://gdal.org/en/stable/api/python/spatial_ref_api.html).

GDAL is an optional dependency to wradlib, whereas `pyproj` is kind of mandatory due to wradlib depending on xradar. So we can make pyproj an official dependency and use this as default our internal projection handling.

Nevertheless we should do the refactor in such way, that users still can rely on the current GDAL based code path, if they provide GDAL OSR objects.

@egouden is already exploring the switch to pyproj in #713. Thanks Edouard for pushing this forward 💯!

egouden, 2025-09-22:
pyproj is much lighter and easier to install than GDAL osr. I think most basic features can be implemented using it.

I suggest to create new functions with pyproj and mark corresponding gdal functions as deprecated, to encourage the switch.

egouden, 2025-09-22:
I understand with your suggestion we actually drop GDAL code. I like it. I will adapt #713 accordingly to see in practice ;-)

closed 2025-09-25 by kmuehlbauer